martymcguire/inkscape-unicorn #7: error

b4zz4, 2015-02-21:
Traceback (most recent call last):
  File "unicorn.py", line 108, in <module>
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 215, in affect
    self.effect()
  File "unicorn.py", line 102, in effect
    parser.parse()
  File "/usr/share/inkscape/extensions/unicorn/svg_parser.py", line 230, in parse
    self.svgWidth = self.getLength('width', 354) * 0.28222
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

martymcguire, 2015-02-21:
Hi b4zz4,

Thanks for reporting this error. There appears to be some element in your SVG that is causing the plugin to choke. Whatever it is appears to have `None` as a width, which is the cause for this error.

Check your SVG for elements that are not groups or paths and try again. I hope this helps!

kobay17, 2015-12-11:
I want to thanks for Marty Mc Guire, for Inscape Unicorn to generate gcode.
I am very interested to learn cnc, for the first one I made cnc machine uses 2 cd rom.
Until now, there has been no lesson to learn cnc in college all campus in INDONESIA "In my country".

Issues that what happens is, when generate Gcode Error. Finally I can find a way to fix the problem. 
I will create a tutorial on youtube how to resolve the issue, later I will update for video tutorial.

MarcoAlici, 2016-05-12:
Same error here. The SVG has no "None", so I think it's a bug.
Hope it helps!
M.

tisaconundrum2, 2016-06-06:
definitely a bug, after altering the parser code. and putting try/except in the code, the error disappeared, so this is definitely a problem with inkscape's API

MarcoAlici, 2016-06-06:
I found that the problem occurs when the image dimension is in not in px. If the dimension is in px it works.
Hope it helps.

svemulapalli96, 2016-12-06:
Hello ! I am also getting the same error as mentioned above. Unable to set the error that encountered.

Traceback (most recent call last):
 File "unicorn.py", line 108, in 
 e.affect()
 File "/usr/share/inkscape/extensions/inkex.py", line 215, in affect
 self.effect()
 File "unicorn.py", line 102, in effect
 parser.parse()
 File "/usr/share/inkscape/extensions/unicorn/svg_parser.py", line 230, in parse
 self.svgWidth = self.getLength('width', 354) * 0.28222
 TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

I am working for my college project. So please do help .

martymcguire, 2016-12-06:
I @svemulapalli96. Please make sure that:

1. You are using the latest version of the extension from the master branch here in GitHub. It contains an improvement that supports more dimension settings on documents.
2. Please make sure your SVG document has units assigned in Document Properties. You should be able to use px, mm, or %.

javeza, 2017-02-25:
The plugin is not compatible with the latest version (0.921 r15371) of Inkscape (as of Feb 25 2017)
I'm getting the exact same error.

martymcguire, 2017-02-25:
Hi Javeza,

Are you seeing this error even when your document has its default units set to px?

javeza, 2017-02-25:
Hello Marty,
When the default units is set to px it works on both 0.92 and 0.91 versions of Inkscape. Does not work when set to mm.

martymcguire, 2017-02-26:
Hi javeza,

I'm glad it is working for you in `px` units. Unfortunately, only `px` units are supported at this time.

closed 2017-02-26 by martymcguire