martymcguire/inkscape-unicorn #8: Newb is seeing an error

Calamarian, 2015-06-18:
When I go to save a file the following error is displayed:

Traceback (most recent call last):
  File "unicorn.py", line 108, in <module>
    e.affect()
  File "D:\Program Files\Inkscape\share\extensions\inkex.py", line 268, in affect
    self.effect()
  File "unicorn.py", line 102, in effect
    parser.parse()
  File "D:\Program Files\Inkscape\share\extensions\unicorn\svg_parser.py", line 230, in parse
    self.svgWidth = self.getLength('width', 354) * 0.28222
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

martymcguire, 2015-06-19:
Thanks for the bug report!

Something changed with the plugin APIs in the latest versions of Inkscape. Unfortunately, I haven't had time to dig in and see what changed.

So, this plugin currently only works with older versions of Inkscape. I don't know when I'll be able to make time to fix this up, but I will happily accept pull requests from anybody who has time to figure this out!

kongorillaSJK, 2015-06-30:
I received this same error with every version of Inkscape I tried. Then I changed my document so that its units are in pix rather than mm. After that, inkscape-unicorn saved the gcode without errors, even from the current Inkscape (v0.91).

garrettx, 2015-09-09:
@kongorillaSJK  Thank you, it works for me

kobay17, 2015-11-27:
Some error, i using Inscape v0.91

Traceback (most recent call last):

  File "unicorn.py", line 108, in <module>

    e.affect()

  File "C:\Program Files\Inkscape\share\extensions\inkex.py", line 268, in affect

    self.effect()

  File "unicorn.py", line 102, in effect

    parser.parse()

  File "C:\Program Files\Inkscape\share\extensions\unicorn\svg_parser.py", line 230, in parse

    self.svgWidth = self.getLength('width', 354) * 0.28222

TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

nouyang, 2016-01-15:
@kongorillaSJK Thanks!
@kobay17   Simply go to File > Document properties and make sure under "Page size, custom size" the units is set to "px" and not "mm". (It is fine for default units to be "mm").

Spankyty, 2016-03-18:
If you need to convert your PX to CM or MM since you cant save as MM or CM I used this website to help:

[http://www.unitconversion.org/typography/pixels-x-to-centimeters-conversion.html](url)

I hope the issue gets fixed soon! :)

chipmunk03, 2016-04-22:
In the code in svg_parser.py, you use 'width' and 'height'. Possibly this has changed in the new version of Inkscape. The new version uses 'Width:' and 'Height:'. I changed that in svg_parser.py, and now it accepts mm. But I don't know if the code is converting px to mm, though. Since in 'Document Properties' one can choose form a list of formats, and in your code you only have options for %, None or px. I doubt it will be saved as mm.

Cheers,

chipmunk03

martymcguire, 2016-04-28:
Thanks for the update! The parser is in desperate need of an update since Inkscape has changed the internals over the last couple of versions.

The G-Code generated by this extension is "unitless", but generally it's assumed that the output is in `mm`. I generally keep my units in `px` with the understanding that the output G-Code moves will be in `mm`.

I don't have time to tackle this any time soon, but I'd like to get some test cases with different units that run against the most recent version of Inkscape, then get the `parseLengthWithUnits` function updated (and supporting the various units).

chipmunk03, 2016-04-28:
Hy Marty,

I discovered that, indeed. But, since the page to set the document properties probably also has changed, the code did not recognise the strings for ‘Width” en “Length”. That was easy to repair. But changing the metrics is a little bit more difficult. 

I’ll dive in to that maybe later. Know I am working on G-Code interpreter for a plotter.

Cheers,

Frank

Virusoft, 2017-03-06:
thanks @nouyang , problem solved.
how much men wrote about but every thing and think is unusable.

Guillaume-slize, 2017-12-19:
Hey,
Is there any news on the development of this. I would like to use the extension with the tinyCNC plotter project. Is there any alternative?
Thanks!

martymcguire, 2017-12-19:
Hi @Guillaume-slize,

I no longer use the (extremely specific) hardware for which I wrote this extension. It was never meant for other CNC machines, despite having been adopted by some projects.

It seems that folks have been having some success using `px` units, or using an older (0.48.x) version of Inkscape.

I would be happy to pass this off to a new maintainer who is focused on the current machines and tools that it is commonly used with.