martymcguire/inkscape-unicorn #13: can't save as gcode

anandvpranavam, 2016-11-12:
it says that "File C:\Users\pranavam\Desktop\dr
[image]
[image]

awing.gcode could not be saved"

anandvpranavam, 2016-11-12:
please help me to save it in gcode format in inkscape.
its for my project.
i am totally doomed by this problem

martymcguire, 2016-11-12:
Hi anandvpranavam,

Sorry to hear that you are having trouble with the plugin.

You may want to try converting the object to a path before saving as G-Code. The plugin doesn't know how to handle every type of object in Inkscape.

Hope that helps!

AMKDS, 2016-12-27:
Please help me: the same anandvpranavam's problem.
You can write mode detail for your solution?

martymcguire, 2016-12-27:
Hi AMKDS,

Sorry to hear that you are also experiencing this issue. Without knowing more about your drawing, I am afraid that the most detailed advice I can give is to make sure to convert all objects in your drawing to Paths before attempting to save as G-Code. You can do this by selecting all objects in your drawing, then using the **Path | Object to Path** menu option.

For (a little) more information, please see the [Converting to path](https://inkscape.org/en/doc/advanced/tutorial-advanced.html) section of the Inkscape documentation.

Hope that helps!

zirolu, 2017-01-02:
i have a same problem, i cant save to .gcode
[image]
[image]

martymcguire, 2017-01-02:
Hi @zirolu. This appears to be a different issue. The error you are seeing occurs when your document has no units. Make sure that your document has units set in px or mm.

martymcguire, 2017-01-02:
Closing this issue as I haven't heard any updates in a while.

zirolu, 2017-01-03:
in document properties i set units to mm. but still can't save to .gcode
[image]

zirolu, 2017-01-03:
can you give me some tutorial. because i am very very newbie. thankz

martymcguire, 2017-01-03:
hi zirolu,

Are you still seeing the same error (`unsupported operand type(s) for *: 'NoneType' and 'float'`)? Or a different error?

zirolu, 2017-01-03:
I see the same error..

>
> hi zirolu,
>
> Are you still seeing the same error (unsupported operand type(s) for *:
'NoneType' and 'float')? Or a different error?

martymcguire, 2017-01-03:
The plugin is only capable of handling some SVG objects and it works best with paths. If you have text or other complex objects, it can help to convert them to paths before trying to save as G-Code.

zirolu, 2017-01-04:
First..i draw an object.. i set unit in milimeters..
Then i path to gcode.. next i save to .gcode but that error appear, so i
cant save..
What should i do???

martymcguire, 2017-01-04:
Hi zirolu,

I am confused by your description of the process. Can you try following these steps?

1. Create a new blank drawing.
2. Set units to millimeters.
3. Draw a shape with the pencil tool.
4. Save the file as SVG.
5. Try to save as G-Code.

If the plugin fails to save as G-Code, please include your SVG file in the reply.

zirolu, 2017-01-04:
Sorry...my english is not quite well..
Ok. I will try..

zirolu, 2017-01-04:
[image]
1. new blank drawing

zirolu, 2017-01-04:
[image]
2. set unit to mm

zirolu, 2017-01-04:
[image]
3. drawing a shape

zirolu, 2017-01-04:
[image]
4. save as .svg

zirolu, 2017-01-04:
5. save as .gcode
[image]
[image]

zirolu, 2017-01-04:
6. Still problem
[image]
[image]

martymcguire, 2017-01-04:
Thank you for the screenshots! I really appreciate that you have given me a complete view of the process.

What happens if you try setting the units to "px" instead of "mm" before generating G-Code?

zirolu, 2017-01-04:
There is no problem to saving when i set unit in px.
Thank you...

fabriciokjc, 2017-01-06:
I have the same problem while using inkscape, I can not save the file in 'mm'. Saving in unit 'px' worked.

sarajitmitra92, 2017-01-28:
[image]
 i am facing this problem please help me

martymcguire, 2017-01-28:
hi @sarajitmitra92,

this looks like a different problem to the above. please make sure that you have copied all of the `unicorn.inx`, `unicorn.py`, and `unicorn/` folder to your Inkscape's extension folder.

sarajitmitra92, 2017-01-28:
from where i can get this files?????plz send me links

martymcguire, 2017-01-28:
They are here in this GitHub repository. You can download all of the files in a .zip archive here:

https://github.com/martymcguire/inkscape-unicorn/archive/master.zip

Then, follow the instructions in the [Installation section of the README](https://github.com/martymcguire/inkscape-unicorn#install).

Good luck!

paswan, 2017-01-31:
sorry i have to say that still i am getting problem.........the problem is same as the problem got by zirolu.......

martymcguire, 2017-01-31:
Hi @paswan - please make sure your drawing unix are in `px` before saving. the extension does not support other units.

iyyanar312, 2017-02-01:
[image]
sir, why this option hidden. i have to know that reson . plz help me

iyyanar312, 2017-02-01:
bcz my arduino based cnc machine doesn't move Y axis

martymcguire, 2017-02-01:
Hi iyyanar312. I am sorry to say that I do not understand your question.

If you are looking to set your document's units to "px", you are very close to the correct option:

[image]

Can you help me with more information to understand the issue with your machine's Y axis? Does it move normally with G-Code produced by other tools?

yasin619, 2017-02-05:
I have a problem while saving g code I've gone through all the descriptions please help me

yasin619, 2017-02-09:
Yes, iam getting error at the end while saving

martymcguire, 2017-02-09:
Hi @yasin619,

Please check your document properties and make sure the units are set to "px". The extension does not support other units.

yasin619, 2017-02-10:
Thank you so much

tron9999, 2017-02-28:
Hi there,
I have the same problem as iyyanar312,
Can only save in px
Cnc maschine is set to mm
Files saved in px cause y axis not to work
There is also a problem if you type 50 mm for width then inkscape enters this as 50,000
Please help

martymcguire, 2017-02-28:
Hi tron9999,

Sorry to hear that you are having trouble.

When your document is in `px` units by default, you must work with `px` units everywhere. So, instead of entering 50mm, try entering 50px.

Can you explain more about the issues you are seeing with the y-axis?

tron9999, 2017-02-28:
Hi there thanks for your swift reply. I have tested my cnc with other files that people made in inscape but set from binging to mm. It works well. I can save files in px in inscape but then the y axis does not work
Regards

tron9999, 2017-02-28:
Sorry need to add I can't save gcode in mm at all, some floating point error, think the problem here is the fact that I can't change my mm to be 50mm instead of 50,000mm cant get rid of the 3 digits at all

martymcguire, 2017-02-28:
Hi tron9999,

Unfortunately the plugin doesn't support SVG documents in mm units at all, only px units.

tron9999, 2017-02-28:
Ok thank you for trying to help me

tron9999, 2017-03-09:
Hi there, Inkscape version 0.48.5 works well with the gcode plugin set to mm  so thank you.👍🏻

media1328, 2017-05-04:
Hi here I have some problem to save in G-code mode as the screen shut say please help my how can i solve my problem
[image]

media1328, 2017-05-04:
and my next screen shut is as below:
[image]
I suggest please try to solve all bugs before distributing such a good software that others can work more easily if possible thanks again

martymcguire, 2017-05-04:
Hi media1328,

That error most often shows up when the extension cannot find some of its files. Please make sure that the files `unicorn.inx` and `unicorn.py`, and the `unicorn` folder (which contains `context.py`, `entities.py`, and `svg_parser.py`) are all copied into your Inkscape extensions folder.

media1328, 2017-05-04:
Hi thanks for your early reply for sure yes all you say are in my extension
folder but the result is the same as before give you one screen shut

​thanks and best regards​

-- 
Good* Love *gives you Good and green* Life*

martymcguire, 2017-05-04:
Hi media1328,

That looks like it should be right. However, the error indicates that for some reason it cannot find the "context.py" file in the unicorn folder. Can you confirm that the three .py files are in that folder and that they are readable?

Thanks

vinodmoorthy, 2017-05-08:
I need a help I get this error message before saving to G-code 
C:\programfiles(x86)\inkscape\pythom\pythom.exe can't find _main_module in '.'

martymcguire, 2017-05-09:
Hi vinodmoorthy,

I'm sorry to hear that you are having trouble. I am afraid that I have not seen that particular error before.

You might want to try re-installing Inkscape, then installing the extension, making sure to copy the unicorn.inx, unicorn.py, and unicorn/ folder into the extensions folder of your fresh installation of Inkscape.

vinodmoorthy, 2017-05-09:
Hi Marty,

Thanks I got it running by recopying the src files back and changing the setting to pk 

Thanks for your help once again 

Regards 
Vinodmoorthy

media1328, 2017-05-09:
Hi yes all you said I have done but nothing happens for me all as the same I did before. yes really I reinstall software more than two times. really I got tired and forget it to do any more and that's why I didn't replay it to you sorry . also all files you say is copied to an extension directory. 
have a good day

martymcguire, 2017-05-09:
Hi media1328,

I am sorry that you are still having trouble. The screenshot of your extensions folder suggests that all the files are in the correct place. However, the error message from Inkscape says it cannot find the correct files.

At this point I am not sure what to try next.

media1328, 2017-05-13:
Hi here is me again. 
for your more sure I sent you some screen shut again that can see and if possible help me to solve my problem really I need this software urgently to use it in my CNC home made for PCB's that I make. for sure there most be away to solve. other's how dose work with this software? and again for sure in can acts well. 
thanks need your help dear   Martymcguire

[image]
[image]
[image]

martymcguire, 2017-05-13:
Hi media1328,

From the screenshots, it looks like every file is in the correct place.

Are you still seeing the "ImportError: No module named context" error? Or perhaps a different error?

HaiNguyen0812, 2017-06-07:
Hi Everyone, I have the same problem and i have re-install Inkscape in 0.48.5 version, now i can do it with px or mm.

Chaitnaya, 2017-07-04:
[image]
I am getting this error.

ecesubhajit, 2017-07-04:
Thanks a lot  #martymcguire.

Brothers2000, 2017-07-22:
Thanks, I had the same problem and I could solve it. martymcguire

[32 more replies]

closed 2017-11-04 by martymcguire