martymcguire/inkscape-unicorn #14: can't save as g.code... why?

versphere, 2016-12-11:
[image]

versphere, 2016-12-11:
why i get this error when i want to save as g.code

martymcguire, 2016-12-12:
HI versphere,

First, please make sure that you fully followed the installation instructions. That error usually indicates that some of the files for the extension are missing. You need to copy the entire contents of the `src/` folder to your Inkscape's extensions folder, including unicorn.inx, unicorn.py, and the unicorn/ folder with all of it's files.

MrFranks, 2016-12-12:
I have tried this on the 32 and 64bit i get the same error on both applications. I followed the install instruction ant least 12 time now. Any ideas?

Traceback (most recent call last):

  File "unicorn.py", line 108, in <module>

    e.affect()

  File "C:\Program Files (x86)\Inkscape\share\extensions\inkex.py", line 268, in affect

    self.effect()

  File "unicorn.py", line 102, in effect

    parser.parse()

  File "C:\Program Files (x86)\Inkscape\share\extensions\unicorn\svg_parser.py", line 230, in parse

    self.svgWidth = self.getLength('width', 354) * 0.28222

TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

martymcguire, 2016-12-12:
Please make sure that your SVG file has units set in the Document Properties. px or mm should work.

MrFranks, 2016-12-12:
Thanks for the quick reply but its still a no, I  also read that fix on the thingiverse forum.[http://www.thingiverse.com/thing:5986/#comments]  Ah well I will try a full install again tomorrow and see if I can get it to work. please let me know if you think of another way around this darn error. 

Thanks for your time

MrFranks, 2016-12-13:
Uninstalled and started clean with 32 bit. Worked first try this morning :). Also made sure settings where set to px for measurements.
Thanks for the help.

martymcguire, 2016-12-13:
Excellent! Glad you got it working. I'll close this issue, now.

closed 2016-12-13 by martymcguire