martymcguire/inkscape-unicorn #15: Only one of my x-axis moves

ballethorsoe, 2017-04-19:
Hi,
Im doing a project at school and our cnc works fine when other people make a .gcode drawing as path but when i make one, only the x-axis moves. ( https://www.youtube.com/watch?v=2VFOU-WUQIY&t ) This is how we made our cnc. Please get back to me as quick as possible :-)

martymcguire, 2017-04-19:
Hi ballethorsoe,

It may help to compare your Inkscape extensions directory with other students who are generating working G-Code.

Another thing to try is to make sure that your document units for your SVG are set to "px".

Unfortunately I don't know how much I can help. This extension was only designed to be used with a Cupcake CNC with the Unicorn drawing attachment.

ballethorsoe, 2017-04-20:
We got it working. Thanks :)

closed 2017-04-20 by martymcguire