martymcguire/inkscape-unicorn #16: can't save as gcode : no. 2

J-Dunn, 2017-04-28:
I have just tried to install the extension on a Fedora system with Inkscape 0.92

After unzipping I changed all the files to 755 permissions to make them executable and copied them to the inkscape extensions location. 

I drew a simple free hand line and saved as SVG. OK
I did Path | Stroke to path , then tried to save a copy:  no gcode extension listed. 
I named with a .gcode extension and selected "guess from extension" : it said it did not recognise this file type. 

I guess the extension is not loading or produces some silent error which Inkscape does not bother reporting to me. 

Can you help?
TIA.

martymcguire, 2017-04-29:
Hi J-Dunn,

If the Save a Copy dialog doesn't show "MakerBot Unicorn G-Code (*.gcode)"
as an option, it sounds like Inkscape isn't finding the files for the
extension.

If it helps, here's a partial contents listing of my Inkscape extensions
folder (emphasis added):

...
triangle.py
uniconv-ext.py
uniconv_output.py
*unicorn/*
*unicorn.inx*
*unicorn.py*
voronoi.py
voronoi2svg.inx
...

Good idea checking the file permissions. 0755 seems to match what I have
here.

I hope you are able to get it working!

J-Dunn, 2017-04-30:
many thanks. I think I had not noticed that unicorn was a directory and copied the files with cp instead of cp -a 

Anyway I now have the .gcode extension available.  ;)

martymcguire, 2017-04-30:
Excellent!

Bravobhr, 2017-10-22:
Hi friend. i use win xp. so i can't save to file gcode. can you explein to me. how to make me save objec into file gcode.Thank's

martymcguire, 2017-10-22:
Hi @Bravobhr.

I am not sure, but I believe that the older versions of Inkscape that this extension was written for may still run on Windows XP.

You might have some luck installing [Inkscape 0.48](https://inkscape.org/en/release/0.48.5/platforms/) to use with this extension.

Bravobhr, 2017-10-29:
thank you very much for your knowledge and I think you are a technological
innovation teacher for me. I hope you succeed. 👍😁

Bravobhr, 2017-10-29:
Teacher Sir Marty
I try to make mini cnc with material: 2 dvd motor, sevo, arduino uno,
stepper L239d. I have problem with serial port in program processing does
not appear arduino port connection caption box with processing after
uploaded java gctrl. can you please explain to me with one by one. thank you

2017-10-29 18:45 GMT+07:00 Bravobhr Ham <bravobhr@gmail.com>:

brindabans, 2017-11-04:
inscape shows this msg when i save a image file as a .gcode "Traceback (most recent call last):
  File "unicorn.py", line 108, in <module>
    e.affect()
  File "inkex.py", line 283, in affect
    self.effect()
  File "unicorn.py", line 102, in effect
    parser.parse()
  File "unicorn/svg_parser.py", line 230, in parse
    self.svgWidth = self.getLength('width', 354) * 0.28222
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float' "
plz help me

martymcguire, 2017-11-04:
Hi @brindabans,

This error occurs with Inkscape versions greater than 0.48.

The first thing to try is to make sure your document units are set to "px".

If that doesn't work, try downgrading to Inkscape 0.48 and re-installing the plugin.

closed 2017-11-04 by martymcguire

Bravobhr, 2017-11-06:
Dear M
Thankyou verymuch. i try and it's work, so, next time i want to try make
quadcopter from arduino uno, i wish you help me, 0nemore again thank's for
your lesson