martymcguire/inkscape-unicorn #24: problem with position

georgeraveen, 2017-11-30:
i created gcode file
but the position (0,0) in the inkspace is  >>>   X-12.09 Y19.13 in the gcode file
how to fix this
please help

martymcguire, 2017-12-04:
hi @georgeraveen,

Sorry to hear that you are having trouble. I have not encountered this particular issue, so I am not sure what might be causing it.

Perhaps you could translate your entire document by X+12.09 Y-19.13 and see if (0,0) comes out at (0,0) again.

I hope that helps!

georgeraveen, 2017-12-04:
@martymcguire  **thank you for replying me.**

what's translating in gcode.? how to do that? 

**also**

I noticed that the 0,0 position of gcode differs according to the canvas size of the inkspace.
for a better understand i attached some screenshots

**here is a canvas of 300*300mm and a rectangle drawing**
 
[image]

[image]

watch these out and let me know if you got any solution.

**thanks you so much.**

martymcguire, 2017-12-04:
hi @georgeraveen. Out of curiosity, does the g-code *end* at X0.0 Y0.0?

georgeraveen, 2017-12-05:
Hi  @martymcguire   
Yes , what ever the canvas matters, gcode starts ends at(0,0).

Inkspace knows where the zero point is situated but it doesn't place the image on zero point.
User has to find the (0,0) point on inkspace and place the image which is out of canvas.

Thank you for helping me