martymcguire/inkscape-unicorn #27: Error

tsromail, 2017-12-26:
Hello

First I am new in this issue with Arduino and I done a CNC from dvd's.
Its working fine with downloaded ready done gcodes but I want to use for circuits boards.
When I try to compile .gcode in Inkscape, I have an error:
" Traceback (most recent call last):
  File "unicorn.py", line 22, in <module>
    from unicorn.context import GCodeContext
  File "C:\Program Files (x86)\Inkscape\share\extensions\unicorn\context.py", line 1
    1 from math import * 
         ^
SyntaxError: invalid syntax

Can anyone help with this issue please.
All the best
tsro

martymcguire, 2017-12-26:
Hi @tsromail,

I have not seen this error before. What version of Inkscape are you using?

The last version this plugin really worked well with was 0.48.5.

tsromail, 2017-12-27:
Hello
Thank you for your prompted response.
After many trays I found and installed ver. 0.48.5 recommended in web page.
I have possibility to request save as .gcode but give that error. 1 from math import *.
All the best
 tsro

martymcguire, 2017-12-27:
Hi @tsromail,

Sorry to hear that you are still having trouble.

From the text of that error, it sounds like perhaps you have copied and pasted the contents for the file from an app that includes line numbers? The file should not contain line numbers.

For comparison, the start of `unicorn/context.py` on my system reads:

```
from math import *
import sys

class GCodeContext:
```

tsromail, 2017-12-28:
Hello Marty

 

Thank you very much for your help, you was right. After I delete a lot of lines number all was ok.

All the best

Stelian Tache

martymcguire, 2017-12-28:
Great! I'm glad that you were able to get it working. Best of luck with your project!

closed 2017-12-28 by martymcguire