martymcguire/inkscape-unicorn #30: issue while generating g-code

neelji, 2018-02-19:
[image]

martymcguire, 2018-02-19:
Hi @neelji,

The error in that screenshot appears to be coming from a "servo.py", which is not part of this extension.

closed 2018-02-19 by martymcguire

neelji, 2018-02-19:
here is the complete message
[image]
can you please explain the solution, actually i am very new user

martymcguire, 2018-02-19:
hi @neelji,

Thank you for the extra details.

Unfortunately, as I said before, this error is occurring in code that is not a part of this project.

I am afraid that I cannot help you here.